aussierobots/ublox_dgnss #1: Failed to find rclcpp on ROS 2 build farm

sloretz, 2021-07-21:
Replying to this comment: https://github.com/ros/rosdistro/pull/30258#issuecomment-883932410

When a package fails to build on the buildfarm, there should be a link to the job that failed on Jenkins. Look for the  "Console Output" link, and then the "Full Log" link on that page if the log is long enough. Read through the log to find the build failure. I find it easiest to find the error by going to the `build binarydeb` section and reading down.

https://build.ros2.org/job/Gbin_uF64__ublox_dgnss_node__ubuntu_focal_amd64__binary/2/consoleFull#console-section-15

In this case the error is:

```
23:04:53 CMake Error at CMakeLists.txt:22 (find_package):
23:04:53   By not providing "Findrclcpp.cmake" in CMAKE_MODULE_PATH this project has
23:04:53   asked CMake to find a package configuration file provided by "rclcpp", but
23:04:53   CMake did not find one.
23:04:53 
23:04:53   Could not find a package configuration file provided by "rclcpp" with any
23:04:53   of the following names:
23:04:53 
23:04:53     rclcppConfig.cmake
23:04:53     rclcpp-config.cmake
23:04:53 
23:04:53   Add the installation prefix of "rclcpp" to CMAKE_PREFIX_PATH or set
23:04:53   "rclcpp_DIR" to a directory containing one of the above files.  If "rclcpp"
23:04:53   provides a separate development package or SDK, be sure it has been
23:04:53   installed.
```

If CMake can't find a package on the buildfarm, it usually means the package being built isn't declaring it as a build dependency in the `package.xml`. The buildfarm installs only the package's dependencies when it tries to build it, so if it doesn't declare `rclcpp` then `rclcpp` won't be installed. That seems to be the case here. It looks like `ublox_dgnss_node` needs a `<depend>rclcpp</depend>` tag so that `rclcpp` is sure to be installed both when the package is built on the buildfarm and when a user installs installs `rclcpp-galactic-ublox-dgnss`.

https://github.com/aussierobots/ublox_dgnss/blob/498f811f4c2d845855ccd01b2331df688bbe18e8/ublox_dgnss_node/package.xml#L13-L20

hortovanyi, 2021-07-21:
@sloretz Ive got a dependencies also to libusb-1.0. Uncertain where to add that?

hortovanyi, 2021-07-21:
Changes applied and new PR created  https://github.com/ros/rosdistro/pull/30296

hortovanyi, 2021-07-21:
Missed rclcpp_components and some message includes - fixed with new PR https://github.com/ros/rosdistro/pull/30299

hortovanyi, 2021-07-22:
@sloretz am a bit lost with why its failing on "/usr/bin/ld: cannot find -lpthreads"

hortovanyi, 2021-07-26:
building https://build.ros2.org/job/Gdev__ublox_dgnss__ubuntu_focal_amd64/8/

closed 2021-07-26 by hortovanyi