aussierobots/ublox_dgnss #3: Failed to build ublox_dgnss_node on Rolling/Humble

clalancette, 2022-05-23:
This package is failing to build on the buildfarm for Rolling and Humble, one example is: https://build.ros2.org/job/Hbin_uJ64__ublox_dgnss_node__ubuntu_jammy_amd64__binary/9/consoleFull .

The problem ends up being this:

```
00:01:22.492 CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
00:01:22.492   Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
```

The containers that the ROS buildfarm uses *only* install things that are listed in the package.xml.  In this case, the package.xml does not list a dependency on `pkg-config`, and so that executable is not available for CMake to find.  If you add a `<build_depend>pkg-config</build_depend>` to your package.xml and do a new release, that should fix this problem.

hortovanyi, 2022-05-23:
It’s exactly the same as I used for galactic. Have no idea what is missing?

hortovanyi, 2022-05-23:
@clalancette its exactly the same as what I’m using to build it on galactic … where is this requirement listed?

hortovanyi, 2022-05-23:
@clalancette i don’t use pkg-config and it compiles/builds successfully on my local environment. This seems to be an build-farm configuration issue?

gavanderhoorn, 2022-05-23:
https://github.com/aussierobots/ublox_dgnss/blob/120c4c89db9e5bf993c9e091da5bce158b537d9e/ublox_dgnss_node/CMakeLists.txt#L28-L29

?

hortovanyi, 2022-05-23:
@clalancette that was to include libusb-1.0 only … works everywhere else. Is libusb-1.0 library available still on the buildfarm virtual machines for humble on Ubuntu jammy?

gavanderhoorn, 2022-05-23:
I would agree with @clalancette: this is more likely a problem with you not stating that dependency in your manifest than with the buildfarm config or whether Jammy introduced some sort of change.

Regardless of whether your package happened to build previously, you should list all your dependencies explicitly. Avoid transitively depending on something, as this can lead to problems with your immediate dependencies change their dependencies.

hortovanyi, 2022-05-23:
@gavanderhoorn it’s a cmake module https://cmake.org/cmake/help/latest/module/FindPkgConfig.html .. libusb-1.0 is the actual dependency.

gavanderhoorn, 2022-05-23:
So CMake tries to use Pkg-Config to find `libusb`, as you ask it to do here:

https://github.com/aussierobots/ublox_dgnss/blob/120c4c89db9e5bf993c9e091da5bce158b537d9e/ublox_dgnss_node/CMakeLists.txt#L28-L29

if Pkg-Config is not present on the system where CMake tries to do that, you'll get this error:

```
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
```

which is the error you find in your buildfarm build log.

So even though your code depends on `libusb`, your *build* additionally depends on `pkg-config`.

If you don't state that dependency, your build will fail.

hortovanyi, 2022-05-23:
@gavanderhoorn @clalancette change made but I cant bloom-release it 
```
==> Checking on GitHub for a fork to make the pull request from...
Could not find a fork of ros/rosdistro on the hortovanyi GitHub account.
Would you like to create one now?
Continue [Y/n]?
Aborting pull request: HTTP Error 401: Unauthorized (https://api.github.com/repos/ros/rosdistro/forks)
The release of your packages was successful, but the pull request failed.
Please manually open a pull request by editing the file here: 'https://raw.githubusercontent.com/ros/rosdistro/master/humble/distribution.yaml'
<== No pull request opened.
```

hortovanyi, 2022-05-23:
Project building successfully now thank you @gavanderhoorn @clalancette for your help

closed 2022-05-23 by hortovanyi