aussierobots/ublox_dgnss #5: Product ID and F9R compatibility

tomlogan501, 2023-03-24:
Hi,
I'm trying your branch F9R and got issues with a custom product ID to differentiate from F9P.
I enforce it in the USB header but it doesn't want it and tell me "usb device not found".
Is there any restriction with the libusb about that ?
I got two udev rules to redirect F9R and F9P before that.
Could you add a parameter to change the productid ?

Thanks.

hortovanyi, 2023-03-26:
Presently it just detects the first device found using libusb ... have been thinking about a way to differentiate. I'm not sure of the correct approach as yet. Someone else wanted to be able to connect 3 x F9Ps to one machine. There is also the issue of the fact that there is different capabilities between the different models. Some messages should be turned off if they arent applicable to the model connected.

tomlogan501, 2023-03-26:
I also want to connect 2 to 3 F9 devices ... we will make a club :) 
Would it be better to switch to /dev/tty mount ?

gsokoll, 2023-03-28:
Tom, the current code looks for devices that match the USB Vendor ID (0x1546) and Product ID (0x01a9) values for the ZED-F9R.  As mentioned, it just detects the first device that matches these using libusb.

These USB ID's are defined on the device itself - using udev rules to change where the device is mounted doesn't change these values.   Switching to /dev/tty* likewise will not change the ID's.

Note that ublox appears to use the same product ID across multiple models of GPS.  As far as I can see, the ZED-F9P, ZED-F9R, and M8 all use 0x01a9 for their product ID.

You can actually change the USB ProductID via the CFG-USB configuration settings - see screenshot below from u-centre.  If you were desperate, you could modify these on your devices and add the product ID as a parameter (look [here](https://github.com/aussierobots/ublox_dgnss/blob/3bb1a564c3a28d24ddbef251e7f4f6bae7a208d7/ublox_dgnss_node/include/ublox_dgnss_node/usb.hpp#L32) in the code).

I'm not sure this is a wise thing to do though, as it may have other unintended consequences.  There is however a 24-byte serial number string available which you can program to whatever you like.  Getting the ublox-dgnss driver to optionally find the matching device using this may be a better approach.

[image]

miguelao33, 2023-09-02:
Hello there,

Is this code functional on a Ardusimple simpleRTK2B receiver?

https://www.ardusimple.com/product/simplertk2b-basic-starter-kit-ip65/

I've just bought it and doesn't support humble..

thanks!

hortovanyi, 2023-09-03:
@miguelao33 I don't have one of those devices. If its configurable via USB through ublox u-center it should work. There was a humble release, however the present driver version is primarily targeted now for ROS2 Iron

gsokoll, 2023-09-04:
@miguelao33 we have an Ardusimple RTK2B working without problems on Humble.  What issues have you encountered?  Have you set a udev rule as mentioned in the readme?  Can you access and configure the device over USB using u-center2?

miguelao33, 2023-09-04:
Hello Geoff,

I had no problem. I just received the ardusimple board and it works at the
moment I connected with your Code perfectly.

The only issue was having a rule applied to the same device (old test),
then the launch file crash and give me the error "cannot connect to usb".

Thanks!

closed 2023-11-07 by hortovanyi