aussierobots/ublox_dgnss #13: Ublox_gnss_node not publishing odometry

miguelao33, 2023-10-10:
Hello there! 

I was trying to check what is inside the odom messages but it shows no info, is not being publish but is declarated, I was trying changing booleans as CFG_ODO_USE_ODO/COG to true but still nothing.

Could you please point me out on how to see the odometry calculated? 

thanks a lot.

hortovanyi, 2023-10-10:
Try setting this parameter CFG_MSGOUT_UBX_NAV_ODO_USB to say 10 either in the launch file, via RQT or command line

miguelao33, 2023-10-10:
Could you give me an example on how to do it in command line? I try this:

ros2 param set /ublox_dgnss CFG_MSGOUT_UBX_NAV_ODO_USB:=10

with the node going. the answer is :the following arguments are required: value

hortovanyi, 2023-10-10:
Have a look at the sample launch files. Need to copy one and set the values for your environment. That will be easier then setting them all with ros2 run

miguelao33, 2023-10-10:
Yes, you are right, better with the launch file. 

I got the /fix publishing but at a rate of 2 messages / minute, or even less. in the other hand, nav_status publishes every 3-4 seconds. 
On my i set those values, is there anything wrong?

  params = [{'CFG_USBOUTPROT_NMEA': False},
            {'CFG_RATE_MEAS': 100},
            {'CFG_RATE_NAV': 1},
            {'CFG_MSGOUT_UBX_NAV_HPPOSLLH_USB': 1},
            {'CFG_MSGOUT_UBX_NAV_STATUS_USB': 5},
            {'CFG_MSGOUT_UBX_RXM_RTCM_USB': 1},
            {'CFG_MSGOUT_UBX_NAV_COV_USB':1},
            {'CFG_MSGOUT_UBX_NAV_ODO_USB':1}]

what can I do to get a publish rate as /ubx_nav_hp_pos_llh

hortovanyi, 2023-10-10:
Geoff created that node to publish NavSatFix. I don't use it myself.

Looking at the code
https://github.com/aussierobots/ublox_dgnss/blob/main/ublox_nav_sat_fix_hp_node/src/ublox_nav_sat_fix_hp_node.cpp￼
ublox_dgnss/ublox_nav_sat_fix_hp_node/src/ublox_nav_sat_fix_hp_node.cpp at main · aussierobots/ublox_dgnss
github.com

It needs to receive a Ubx Nav Status and Ubx Nav Cov then it will allow the next UBX Nav HP Pos LLH to be published as the Nav Sat Fix.

This might be too restrictive. Have noticed that Ubx Nav Status is a lower priority message.

Maybe create a local node and try commenting these lines out where the have_recd_ variables are set to false;
https://github.com/aussierobots/ublox_dgnss/blob/8e72b574aec3d38aa2a1bb73b1927035538979ff/ublox_nav_sat_fix_hp_node/src/ublox_nav_sat_fix_hp_node.cpp#L95￼
ublox_dgnss/ublox_nav_sat_fix_hp_node/src/ublox_nav_sat_fix_hp_node.cpp at 8e72b574aec3d38aa2a1bb73b1927035538979ff · aussierobots/ublox_dgnss
github.com

miguelao33, 2023-10-10:
Great, is publishing now with a good rate. 

I've also check this: [https://stackoverflow.com/questions/63037315/how-to-receive-gps-data-faster-using-kumarrobotics-ublox-package]

Thanks a lot, have a nice one!

hortovanyi, 2023-10-10:
Its a bit of playing around sometimes to get what you want out of the devices .. glad you've got a result :) 

There are a backlog of changes coming through. I might also look at revising some of that code in the UBLOX Nav Sat Fix Hp node such that its not so reliant on updates from the none UBX Nav HP Pos LLH messages

closed 2023-10-10 by hortovanyi