aussierobots/ublox_dgnss #15: Frame ID and data visualization

miguelao33, 2023-10-13:
Hello guys,

sorry to place another idiot issue. :(  

I have all nodes functioning and publishing on /fix topic. I also want to integrate it with nav_sat node from robot localization package, but it seems is not picking the data from the /fix topic. On my research, i cross with "mapviz" package and there, i was also unable to see the /fix messages from gps or from navsat message-types. I'm afraid that the used frame id "ubx" is not allowing me to do that, I dont now if i need to republish them, to make a transform to my base_link or any other possible solution. 

If you were trying to use nav_sat from robot_localization, which steps will you follow? also, there's any other tool to allow me to see the publish data (as to see a test path, for example) ?

Thanks a lot!

hortovanyi, 2023-10-13:
I think you may be coming across a QOS configuration issue. All output presently I believe is using the SensorQOSQuality of Service settings — ROS 2 Documentation: Iron documentationdocs.ros.orgHave a look at thisExternal configurability of QoS policies.design.ros2.orgIt might be that the packages you are using are still using the ros1 qos profile defaults?If you have a very accurate position why do you need robot_localization?

gsokoll, 2023-10-14:
You can visualise the /fix messages using Foxglove Studio.

miguelao33, 2023-10-14:
Hello, 

yeap, foxflove studio is showing from the robot to my computer, it must be something with the navigation2.. 

Well, as the outdoor feel will be problematic, I need to fuse Imu and GNSS. Also, I'll have 2 platforms, 1 that just follow waypoints and other that also navigate (I'll include lidar soon) and I'd like to use the planners from nav2. Also, some day it will be more than 1 robot, so I want all be below the same structure nav2 provides (tasks, etc.)

I'll continue and if tell u guys if I run into another crossroad. Take care, and thanks again!

alistair-english, 2023-10-15:
> I'm afraid that the used frame id "ubx" is not allowing me to do that, I dont now if i need to republish them, to make a transform to my base_link or any other possible solution.

Hi @miguelao33 - just picking up on this, you do need to define a transform from the `ubx` frame back to `base_link` or some other frame on your robot (e.g. via URDF or a static publisher).

Alternatively, you can use the `FRAME_ID` parameter to change the frame that the ublox node publishes messages in.

closed 2024-10-21 by hortovanyi