aussierobots/ublox_dgnss #16: USB poblem with apt instalation

Daniel880, 2023-10-31:
Hi, I have a problem running ZED-F9P when I install this repository from using apt. I get the following error:

`
[component_container_mt-1] component_container_mt: ../../libusb/io.c:1497: libusb_submit_transfer: Assertion 'transfer->dev_handle' failed.
`

[image]

When I clone the repository and build the code myself using colcon, everything works fine.

Can you help me with this problem?

hortovanyi, 2023-10-31:
I just tried using the apt packages and am getting the same problem. It would seem that this is a build farm issue maybe with libusb ... will investigate more 

@gsokoll the last code changes I think stopped throwing an exception in init when dev_handle was a nullptr and returned true/false.. any ideas why this is happening for a libusb transfer?

hortovanyi, 2023-11-01:
@Daniel880 can try setting this environment variable `export LIBUSB_DEBUG=4` and running the command again. Whats the output say?

Daniel880, 2023-11-01:
With LIBUSB_DEBUG=4 there is the following output:

`[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [7681]
[INFO] [component_container_mt-2]: process started with pid [7683]
[component_container_mt-1] [INFO] [1698829716.653679770] [ublox_dgnss_container]: Load Library: /opt/ros/iron/lib/libublox_dgnss_components.so
[component_container_mt-2] [INFO] [1698829716.656022700] [ublox_nav_sat_fix_hp_container]: Load Library: /opt/ros/iron/lib/libublox_nav_sat_fix_hp_components.so
[component_container_mt-1] [INFO] [1698829716.657315625] [ublox_dgnss_container]: Found class: rclcpp_components::NodeFactoryTemplate<ublox_dgnss::UbloxDGNSSNode>
[component_container_mt-1] [INFO] [1698829716.657365261] [ublox_dgnss_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ublox_dgnss::UbloxDGNSSNode>
[component_container_mt-2] [INFO] [1698829716.658406147] [ublox_nav_sat_fix_hp_container]: Found class: rclcpp_components::NodeFactoryTemplate<ublox_nav_sat_fix_hp::UbloxNavSatHpFixNode>
[component_container_mt-2] [INFO] [1698829716.658442099] [ublox_nav_sat_fix_hp_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ublox_nav_sat_fix_hp::UbloxNavSatHpFixNode>
[component_container_mt-2] [INFO] [1698829716.666146194] [ublox_nav_sat_fix_hp]: starting ublox_nav_sat_fix_hp
[component_container_mt-1] [INFO] [1698829716.667155502] [ublox_dgnss]: starting ublox_dgnss
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/ublox_nav_sat_fix_hp' in container '/ublox_nav_sat_fix_hp_container'
[component_container_mt-1] [INFO] [1698829716.677200737] [ublox_dgnss]: Parameter DEVICE_SERIAL_STRING not found, will use first ublox device.
[component_container_mt-1] [INFO] [1698829716.677346835] [ublox_dgnss]: Parameter FRAME_ID not found, defaulting to 'ubx' frame_id
[component_container_mt-1] [INFO] [1698829716.677390943] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_NAV_COV_USB
[component_container_mt-1] [INFO] [1698829716.677413504] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_NAV_HPPOSLLH_USB
[component_container_mt-1] [INFO] [1698829716.677427402] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_NAV_STATUS_USB
[component_container_mt-1] [INFO] [1698829716.677438524] [ublox_dgnss]: parameter supplied: CFG_MSGOUT_UBX_RXM_RTCM_USB
[component_container_mt-1] [INFO] [1698829716.677449016] [ublox_dgnss]: parameter supplied: CFG_RATE_MEAS
[component_container_mt-1] [INFO] [1698829716.677459762] [ublox_dgnss]: parameter supplied: CFG_RATE_NAV
[component_container_mt-1] [INFO] [1698829716.677487165] [ublox_dgnss]: parameter supplied: CFG_USBOUTPROT_NMEA
[component_container_mt-1] [timestamp] [threadID] facility level [function call] <message>
[component_container_mt-1] --------------------------------------------------------------------------------
[component_container_mt-1] [ 0.000028] [00001e20] libusb: debug [libusb_init] libusb v1.0.25.11696
[component_container_mt-1] [ 0.000059] [00001e20] libusb: debug [usbi_add_event_source] add fd 33 events 1
[component_container_mt-1] [ 0.000071] [00001e20] libusb: debug [usbi_io_init] using timer for timeouts
[component_container_mt-1] [ 0.000073] [00001e20] libusb: debug [usbi_add_event_source] add fd 34 events 1
[component_container_mt-1] [ 0.000082] [00001e20] libusb: debug [get_kernel_version] reported kernel version is 6.2.0-36-generic
[component_container_mt-1] [ 0.000122] [00001e20] libusb: debug [op_init] found usbfs at /dev/bus/usb
[component_container_mt-1] [ 0.000125] [00001e20] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[component_container_mt-1] [ 0.000132] [00001e20] libusb: debug [op_init] sysfs is available
[component_container_mt-1] [ 0.000330] [00001e23] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[component_container_mt-1] [ 0.002489] [00001e20] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[component_container_mt-1] [ 0.002497] [00001e20] libusb: debug [linux_get_device_address] scan usb1
[component_container_mt-1] [ 0.002549] [00001e20] libusb: debug [linux_get_device_address] bus=1 dev=1
[component_container_mt-1] [ 0.002554] [00001e20] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[component_container_mt-1] [ 0.002557] [00001e20] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[component_container_mt-1] [ 0.002719] [00001e20] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[component_container_mt-1] [ 0.002725] [00001e20] libusb: debug [linux_get_device_address] scan 1-1
[component_container_mt-1] [ 0.002766] [00001e20] libusb: debug [linux_get_device_address] bus=1 dev=2
[component_container_mt-1] [ 0.002770] [00001e20] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[component_container_mt-1] [ 0.002773] [00001e20] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[component_container_mt-1] [ 0.002823] [00001e20] libusb: debug [linux_get_parent_info] dev 0x7fd0b44fad10 (1-1) has parent 0x7fd0b45155f0 (usb1) port 1
[component_container_mt-1] [ 0.002930] [00001e20] libusb: debug [linux_get_device_address] getting address for device: 1-6 detached: 0
[component_container_mt-1] [ 0.002934] [00001e20] libusb: debug [linux_get_device_address] scan 1-6
[... cut ...]
[component_container_mt-1] [ 0.003756] [00001e20] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x7fd0b44fa8a0 with handle 2
[component_container_mt-1] component_container_mt: ../../libusb/io.c:1497: libusb_submit_transfer: Assertion 'transfer->dev_handle' failed.
[ERROR] [component_container_mt-1]: process has died [pid 7681, exit code -6, cmd '/opt/ros/iron/lib/rclcpp_components/component_container_mt --ros-args -r __node:=ublox_dgnss_container -r __ns:=/'].
`

And two screenshots with program compiled from source and installed with apt. 

apt:
[image]

from source:
[image]

hortovanyi, 2023-11-01:
@Daniel880 thank you for that. Its the same issue I'm facing. Working through it

Are you just trying to connect one USB F9P device?

Daniel880, 2023-11-01:
Only one device. Exactly this board: https://www.sparkfun.com/products/16481

But what is interesting is that when I run compiled code without a plugged device, I get the same error as with the apt version. Additionally, the second observation is that when I use the apt version, there is no difference in output with or without the device plugged.

The third observation is that, when I delete udev rule, there is no difference in apt version output, but in compiled version there is an error about write permission to device.

hortovanyi, 2023-11-01:
@Daniel880 we added changes to be able to specify which usb device to connect to with a usb string. It can be set via u-center.  That way we can determine between 2 or more usb gps devices connected to the one device.

I'm debugging the code now ... the 0.4.4 version doesnt have these changes in it 
https://github.com/aussierobots/ublox_dgnss/releases/tag/0.4.4

hortovanyi, 2023-11-01:
@Daniel880 the last changes I pushed to the repo look promising - my driver is working again

gsokoll, 2023-11-01:
@Daniel880 can you tell us what verion of ROS and OS you're running please?

As @hortovanyi mentioned, the last release added functionality to run multiple instances of the ublox_dgnss node with different ublox devices and this seems to have introduced the issues.  But I have not been able to replicate the errors you see, unless I launch the node without any device attached.

Daniel880, 2023-11-01:
**Ubuntu 22.04.3** with 6.2.0-36-generic kernel
**ROS Iron** version 0.10.0-3jammy.20231010.203644 installed with apt

Now I am using ThinkPad E450 laptop but I also had this problem with my PC, also running the same ubuntu and ros.

hortovanyi, 2023-11-01:
@gsokoll I've worked out what's happening with no device attached - it seems the hot plug attach is causing a transfer to be initiated to the null usb device - that then triggers the assert in libusb which throws the segfault. Not sure how to fix this as yet. It used to throw a red error if no gps usb device was attached.

> But I have not been able to replicate the errors you see, unless I launch the node without any device attached.

hortovanyi, 2023-11-01:
@Daniel880 can you confirm you can use and run the code from the latest source?

hortovanyi, 2023-11-05:
@Daniel880 new release 0.5.2 fixes the problem - apt packages have been updated. Will close this issue if I dont hear anything back to the contrary from you within the next few days. Thank you for raising this.

closed 2023-11-07 by hortovanyi