aussierobots/ublox_dgnss #19: Moving base RTK 问题

noob3-3, 2024-02-22:
Thank you for your project. I am currently trying to use f9p on an outdoor vehicle. I don't quite understand the Moving Base and Rover part. Do I need to launch ublox_mb+r_base.launch.py on the fixed base station and ublox_mb+r_rover.launch.py on the rover station? However, shouldn't there be two rover stations for the vehicle to have heading information? Do I need to modify the parameters and launch ublox_mb+r_rover.launch.py twice? But how do two rover stations calculate heading information?

gsokoll, 2024-02-22:
The "base" device is also mounted to the rover, which is why it is referred to as a "moving base" configuration.  Having both the moving base and the rover devices mounted to the rover allows calculation of heading information.  

If you haven't already, suggest having a read of the ublox document "ZED-F9P-MovingBase_AppNote_UBX-19009093" for detailed guidance.

noob3-3, 2024-02-23:
Thank you for your answer.

If this is the case, then how do the data of two launches on the mobile base and rover device affect each other and ultimately calculate the course?  No other `launch `are started, for example, the position difference between the two devices is fixed, and based on this difference, the positioning data is constrained.

gsokoll, 2024-02-25:
@noob3-3 my apologies, the example launch files for moving base+rover configs include many configuration parameters which the ublox_dgnss driver currently does not support.  These devices should be configured using u-center instead.  I will try to prepare a readme file this week.

There are two devices mounted to the rover, which allows calculation of heading information.  One of them is considered a base station, even though it physically moves (hence the name "moving base").  The app note provides a good explanation of this with several examples.

noob3-3, 2024-02-26:
I am also new to relevant work, thank you for your patient answer  :smile:

closed 2024-03-11 by hortovanyi