aussierobots/ublox_dgnss #21: msg_used time to time jump to 1 from 2

GPrathap, 2024-02-29:
I was running [ublox_rover_hpposllh_navsatfix.launch.py](https://github.com/aussierobots/ublox_dgnss/blob/main/ublox_dgnss/launch/ublox_rover_hpposllh_navsatfix.launch.py) with an NTRIP Client.  I get /fix reading. However, 

`ros2 topic echo /ubx_rxm_rtcm` 

msg_used shows 2, but from time to time msg_used shows 1. Could it have something to do with frequency?  I used the default parameters. Are there any parameters that I should look for?

GPrathap, 2024-03-01:
After checking this https://github.com/aussierobots/ublox_dgnss/blob/main/ublox_ubx_msgs/msg/UBXRxmRTCM.msg#L10 

Is it okay to publish https://github.com/aussierobots/ublox_dgnss/blob/main/ublox_nav_sat_fix_hp_node/src/ublox_nav_sat_fix_hp_node.cpp#L124 only if `msg_used` is equal to 2? Or is this bug in the GPS device?

hortovanyi, 2024-03-01:
Different purposes. One shows that the ublox device received and used a RTCM message. The other publishes the old NavSatFix message using the UBX messages published for the node