aussierobots/ublox_dgnss #28: rtcm_client fails silently if corrections server pauses

PaulBouchier, 2024-10-20:
I am having trouble with version (0.5.2-1jammy.20240731.053557) of ros-humble-ublox-dgnss installed from apt on Ubuntu 22.04 on Raspberry Pi and operating in RTK-GPS mode with an rtk2go.com corrections server. Frequently, at random intervals, the /ubx_nav_hp_pos_llh message field h_acc starts increasing and goes up to 2 meters or so, and differential corrections are lost. Upon investigation, I found /ntrip_client/rtcm would stop sending corrections coincident with the increase in the h_acc field - there were just no more corrections emitted by the rtcm_client, but no error on the terminal in which the client was launched with ros2 launch ublox_dgnss ntrip_client.launch.py use_https:=false host:=rtk2go.com port:=2101 mountpoint:=VN1 username:=paul.bouchier-at-gmail-d-com password:=Unused. Everything looked normal from the terminal and logging point of view. 

I installed ros-humble-rtcm-client to get a second view of what was happening, and it printed the following to the terminal, coinicident with this package's rtcm_client going dead:
Output from ros-humble-rtcm-client shown.
```
[ntrip_ros.py-1] [INFO] [1729373617.877834469] [ntrip_client]: Connected to http://rtk2go.com:2101/VN1
[ntrip_ros.py-1] [ERROR] [1729373845.189211479] [ntrip_client]: RTCM data not received for 4 seconds, reconnecting
[ntrip_ros.py-1] [ERROR] [1729373845.199049326] [ntrip_client]: Unable to connect socket to server at http://rtk2go.com:2101/
[ntrip_ros.py-1] [ERROR] [1729373845.202253430] [ntrip_client]: Exception: [Errno -2] Name or service not known
[ntrip_ros.py-1] [ERROR] [1729373845.205088426] [ntrip_client]: Reconnect to http://rtk2go.com:2101/ failed. Retrying in 5 seconds
[ntrip_ros.py-1] [ERROR] [1729373850.220886096] [ntrip_client]: Unable to connect socket to server at http://rtk2go.com:2101/
[ntrip_ros.py-1] [ERROR] [1729373850.226425367] [ntrip_client]: Exception: [Errno -2] Name or service not known
[ntrip_ros.py-1] [ERROR] [1729373850.229408732] [ntrip_client]: Reconnect to http://rtk2go.com:2101/ failed. Retrying in 5 seconds
[ntrip_ros.py-1] [ERROR] [1729373855.246174099] [ntrip_client]: Unable to connect socket to server at http://rtk2go.com:2101/
[ntrip_ros.py-1] [ERROR] [1729373855.250718507] [ntrip_client]: Exception: [Errno -2] Name or service not known
[ntrip_ros.py-1] [ERROR] [1729373855.254271349] [ntrip_client]: Reconnect to http://rtk2go.com:2101/ failed. Retrying in 5 seconds
[ntrip_ros.py-1] [ERROR] [1729373860.262685469] [ntrip_client]: Unable to connect socket to server at http://rtk2go.com:2101/
[ntrip_ros.py-1] [ERROR] [1729373860.265917054] [ntrip_client]: Exception: [Errno -2] Name or service not known
[ntrip_ros.py-1] [ERROR] [1729373860.269391490] [ntrip_client]: Reconnect to http://rtk2go.com:2101/ failed. Retrying in 5 seconds
[ntrip_ros.py-1] [INFO] [1729373866.772209706] [ntrip_client]: Connected to http://rtk2go.com:2101/VN1
```

This happened again about 5 minutes later, and I believe what's happening is the rtk2go.com ntrip server is pausing for 15-20 seconds every few minutes. This throws the ublox_dgnss rtcm_client for a loop and it doesn't attempt to reconnect, so no corrections are ever sent again until rtcm_client is re-started.

By comparison, ros_humble-rtcm-client times out and retries until it succeeds when the server comes back 20 seconds or so later.

I re-mapped topics to feed the ros-humble-rtcm-client corrections into ublox-dgnss node, and then I got stable operation. I'm unstuck for now, but clearly this is an undesirable state of affairs. I think this rtcm_client needs to detect loss of corrections and retry until it succeeds and be more noisy about the failure.

PaulBouchier, 2024-10-20:
BTW, this happened with two different mountpoints on rtk2go.com. So it's probably not just Joe Blow's base station - it's a broader resilience issue.

hortovanyi, 2024-10-20:
Reading rtk2go website it says it'll email you if something happens. Did you receive any emails?

This Ntrip client works with the auscors network in Australia and with the French/european Ntrip servers.

The error message seems to indicate that you are loosing networking??

hortovanyi, 2024-10-20:
Another option is to compile the code with uncommenting the line curl_easy_setopt(handle, CURLOPT_VERBOSE, 1L); 

In essence libcurl is being used. Any errors should be logged. 

You might also try setting the log level to debug in the launch file.

hortovanyi, 2024-10-20:
@PaulBouchier could you please try setting this in the code with the verbose option https://curl.se/libcurl/c/CURLOPT_MAXAGE_CONN.html 

The suggestion is based on reading this https://stackoverflow.com/questions/70657452/libcurl-checking-the-server-shutdown-connection

PaulBouchier, 2024-10-21:
Hi - thanks for your prompt and effective response.

Working through this, finally got it working like it was yesterday, only using a freshly cloned & built ublox_dgnss package. Answering a couple of questions, with additional testing per your request:

- I did not receive any email from rtk2go. However I did somehow manage to get put in a sandbox for a while after fooling with it this morning.

- My earlier assertion that the ublox_dgnss ntrip_client stops outputting messages on /ntrip_client/rtcm is definitely wrong, as is the assertion that it's always correlated with a dropout as observed and retried by the ros-humble-ntrip_client package. I'm sorry for the wrong data. What I can say is the UBX_NAV_STATUS message field diff_corr goes false frequently and seemed difficult at times to recover using the dgnss ntrip_client.

- I made the configuration changes you requested to ntrip_client.cpp  in the fresh clone as shown below:
```
diff ./ntrip_client_node.cpp ./ntrip_client_node.cpp_MODS
85c85,89
<       // curl_easy_setopt(handle, CURLOPT_VERBOSE, 1L);
---
> 
>       // FIXME: Remove these changes when done with investigation
>       curl_easy_setopt(handle, CURLOPT_VERBOSE, 1L);
>       curl_easy_setopt(handle, CURLOPT_MAXAGE_CONN, 15L);
```

PaulBouchier, 2024-10-21:
It resulted in the client printing the following error every few seconds to the terminal:
```
ros2 launch ublox_dgnss [ntrip_client.launch.py](http://ntrip_client.launch.py/) use_https:=false [host:=rtk2go.com](http://host:=rtk2go.com/) port:=2101 mountpoint:=VN1 username:=paul.bouchier-at-gmail-d-com password:=Unused
[INFO] [launch]: All log files can be found below /home/bouchier/.ros/log/2024-10-20-14-58-02-503335-mowbot-79965
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [79977]
[component_container_mt-1] [INFO] [1729454283.320367721] [ntrip_client_container]: Load Library: /home/bouchier/ublox_dgnss_ws/install/ntrip_client_node/lib/libntrip_client_components.so
[component_container_mt-1] [INFO] [1729454283.359122412] [ntrip_client_container]: Found class: rclcpp_components::NodeFactoryTemplate<ublox_dgnss::NTRIPClientNode>
[component_container_mt-1] [INFO] [1729454283.361681646] [ntrip_client_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ublox_dgnss::NTRIPClientNode>
[component_container_mt-1] [INFO] [1729454283.448764970] [ntrip_client]: starting ntrip_client
[component_container_mt-1] [INFO] [1729454283.455830734] [ntrip_client]: ntrip connection url: 'http://rtk2go.com:2101/VN1'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/ntrip_client' in container '/ntrip_client_container'
[component_container_mt-1] *   Trying 2607:7700:0:4:0:2:38f:f351:2101...
[component_container_mt-1] *   Trying 3.143.243.81:2101...
[component_container_mt-1] * Connected to [rtk2go.com](http://rtk2go.com/) (3.143.243.81) port 2101 (#0)
[component_container_mt-1] * Server auth using Basic with user 'paul.bouchier-at-gmail-d-com'
[component_container_mt-1] > GET /VN1 HTTP/1.1
[component_container_mt-1] Host: [rtk2go.com:2101](http://rtk2go.com:2101/)
[component_container_mt-1] Authorization: Basic cGF1bC5ib3VjaGllci1hdC1nbWFpbC1kLWNvbTpVbnVzZWQ=
[component_container_mt-1] User-Agent: NTRIP ros2/ublox_dgnss
[component_container_mt-1] Accept: */*
[component_container_mt-1] 
[component_container_mt-1] * Failure writing output to destination
[component_container_mt-1] * Closing connection 0
[component_container_mt-1] * Hostname [rtk2go.com](http://rtk2go.com/) was found in DNS cache
[component_container_mt-1] *   Trying 2607:7700:0:4:0:2:38f:f351:2101...
[component_container_mt-1] *   Trying 3.143.243.81:2101...
[component_container_mt-1] * Connected to [rtk2go.com](http://rtk2go.com/) (3.143.243.81) port 2101 (#1)
[component_container_mt-1] * Server auth using Basic with user 'paul.bouchier-at-gmail-d-com'
[component_container_mt-1] > GET /VN1 HTTP/1.1
[component_container_mt-1] Host: [rtk2go.com:2101](http://rtk2go.com:2101/)
[component_container_mt-1] Authorization: Basic cGF1bC5ib3VjaGllci1hdC1nbWFpbC1kLWNvbTpVbnVzZWQ=
[component_container_mt-1] User-Agent: NTRIP ros2/ublox_dgnss
[component_container_mt-1] Accept: */*
[component_container_mt-1] 
[component_container_mt-1] * Failure writing output to destination
[component_container_mt-1] * Closing connection 1
[component_container_mt-1] * Hostname [rtk2go.com](http://rtk2go.com/) was found in DNS cache
[component_container_mt-1] *   Trying 2607:7700:0:4:0:2:38f:f351:2101...
[component_container_mt-1] *   Trying 3.143.243.81:2101...
[component_container_mt-1] * Connected to [rtk2go.com](http://rtk2go.com/) (3.143.243.81) port 2101 (#2)
[component_container_mt-1] * Server auth using Basic with user 'paul.bouchier-at-gmail-d-com'
[component_container_mt-1] > GET /VN1 HTTP/1.1
[component_container_mt-1] Host: [rtk2go.com:2101](http://rtk2go.com:2101/)
[component_container_mt-1] Authorization: Basic cGF1bC5ib3VjaGllci1hdC1nbWFpbC1kLWNvbTpVbnVzZWQ=
[component_container_mt-1] User-Agent: NTRIP ros2/ublox_dgnss
[component_container_mt-1] Accept: */*
```
I don't know if that's significant, but I haven't been able to get this freshly-cloned ntrip_client to make diff_corr go true. That's different from yesterday with the dgnss ntrip_client from ros-humble-ublox-dgnss, which would make diff_corr go true, then it would go false after a few minutes.

PaulBouchier, 2024-10-21:
- I think the freshly cloned dgnss ntrip_client is doing something bad to [rtk2go.com](http://rtk2go.com/) and causing it to put me in jail for a timeout, as evidenced by the following error emitted by ros-humble-ntrip-client - I've been put in the sandbox for maybe a half hour. :-1: 
```
os2 launch ntrip_client [ntrip_client_remap_launch.py](http://ntrip_client_remap_launch.py/) [host:=rtk2go.com](http://host:=rtk2go.com/) port:=2101 mountpoint:=VN1 username:=paul.bouchier-at-gmail-d-com password:=Unused
[INFO] [launch]: All log files can be found below /home/bouchier/.ros/log/2024-10-20-15-08-13-251869-mowbot-80123
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ntrip_ros.py-1]: process started with pid [80124]
[ntrip_ros.py-1] [WARN] [1729454894.756789123] [ntrip_client]: Received unknown error from the server. Note that the NTRIP version was not specified in the launch file. This is not necesarilly the cause of this error, but it may be worth checking your NTRIP casters documentation to see if the NTRIP version needs to be specified.
[ntrip_ros.py-1] [ERROR] [1729454894.759465635] [ntrip_client]: Invalid response received from http://rtk2go.com:2101/VN1
[ntrip_ros.py-1] [ERROR] [1729454894.761898497] [ntrip_client]: Response: 200 Ok, its now sandbox time.
```
(ros-humble-ntrip-client was working perfectly before I ran the freshly-cloned version of dgnss ntrip_client.)

- When I launch the freshly cloned ublox_dgnss satellite node and drive corrections into it from ros-humble-ntrip-client the UBX_NAV_STATUS almost immediately reports diff_corr true and h_acc starts decreasing. This does not happen when I drive corrections from the dgnss ntrip_client - there's no evidence it's using those corrections. Further evidence that the ntrip_client is just not working right with [rtk2go.com](http://rtk2go.com/). I can't get the freshly-cloned ublox_dgnss node to emit UBX_RXM_RTCM, despite setting its config param to 1.

- How can I set the log level to debug from inside a launchfile that uses containers? I can't find any documentation on that.

- Yesterday there may have been some loss at times on the network, because I did see pauses in ssh responses, and I was driving it around. Today it has been sitting stationary and the cell phone with hotspot that connects it to the internet has been right beside the Raspberry Pi - so a more stable environment.

hortovanyi, 2024-10-21:
@PaulBouchier I've been told by another that rtk2go.com is not reliable. There are a few other CURLOPT_ settings to do with connections that might help. However you are going to have to try them out and see if they make a difference.

If you can let me know which ones, then I'll look into settings up parameters for the next version.

I may not have published a humble version of the driver for a while. The only other change I can think of is    curl_easy_setopt(handle, CURLOPT_USERAGENT, "NTRIP ros2/ublox_dgnss"); was added otherwise it doesnt work with the French ntrip server. Maybe try commenting that out. If it does make a difference, again let me know, and I'll add a parameter and change it so its conditionally set such that you can disable it in the launch file

To set the log level have a look here https://robotics.stackexchange.com/questions/99332/specify-log-level-on-ros2-node-in-launch-file

hortovanyi, 2024-10-21:
There might also be another issue with using rtk2go - we may need to set headers to tell it to use Ntrip 2 ?? https://gitlab.com/gpsd/gpsd/-/issues/191

PaulBouchier, 2024-10-21:
Thanks for thinking about this, Nick. I think, considering I have a November 23 deadline, I'm going to put this investigation on hold until after then, and use your ublox-dgnss satellite node with the ros-humble-ntrip-client, which seems to work.

Thanks for the pointer to setting the logging level. I had found that, but hoped you'd been able to use it specifically in the containers you run the nodes in. The example in the link you gave looks like:
```
    Node(
        package="...snip...",
        executable="...snip...",
        output="log",
        exec_name='...snip...',
        arguments=["--ros-args --log-level DEBUG --"],
    ),
```
(with changes as noted in the thread. But your containerized launch file looks a lot different:
```
  container1 = ComposableNodeContainer(
    name='ntrip_client_container',
    namespace='',
    package='rclcpp_components',
    executable='component_container_mt',
    composable_node_descriptions=[
      ComposableNode(
        package='ntrip_client_node',
        plugin='ublox_dgnss::NTRIPClientNode',
        name='ntrip_client',
        parameters=params
      )
```
Maybe the --ros-args should go in the ComposableNodeContainer() block?

Yes, rtk2go.com is unreliable, but it's all we've got in Texas.

Thanks again for looking at this. I think we should leave this ticket open for now, and I can put some more logging in after my deadline and see if I can figure out what's going on. What I don't understand is, diff_corr stays false in the freshly-cloned SW, but ntrip_client is sending corrections data on /ntrip_client/rtcm. Very strange. Maybe a bag file would help.

PaulBouchier, 2024-10-21:
One other thing - and this is not a criticism - I greatly appreciate all the work you did to not just create the node for your own purposes, but then to publish it to a package and answer questions from people like me. 

I understand why you chose to use the SensorMessages Qos profile, it's what OSRF recommends, and for good reason - best effort with minimal delay is the right choice for sensor data. But it comes with downsides that I'm sure you're aware of - can't use rqt_topic, and I'm not yet sure about bagfiles. Essentially, the ROS2 ecosystem has failed to adequately accommodate best-effort Qos. Maybe, as an enhancement, there ought to be a repeater-node that produces "shadow-copies" of the topics except on a RELIABLE Qos topic, for tool support. Or maybe some of the topics that are of a debug nature, like UBX_RMX_RTCM, should be published with a RELIABLE Qos. Something to think about. If you want, I'll make an enhancement request ticket along these lines.

hortovanyi, 2024-10-21:
@PaulBouchier im not really sure what's going on in your environment. You can always use curl command line to figure out the settings for libcurl.

Reading rtk2go it suggests that the default of 5 connections for libcurl might be an issue. Whatever is going on, it is peculiar to rtk2go only and how those servers are set up.

I've posted a few hints in the previous posts to figure it out - if the other script works that's good. 

Yes I might change some of the default libcurl connection settings such that if network connectivity is lost, it'll timeout and reconnect, instead of sitting there silently waiting for the next stream record. 

On the QoS side I've kept the recommended but acknowledge the challenges and yes for ros2 bag you need to override the QoS https://docs.ros.org/en/jazzy/How-To-Guides/Overriding-QoS-Policies-For-Recording-And-Playback.html

I suspect you can do the same in the launch files to. However I might need to change how I create the message publishers https://robotics.stackexchange.com/questions/105844/is-there-a-workaround-to-set-qos-at-launch-time

I just haven't had a need to do this (except overriding the qos for ros2 bags)

hortovanyi, 2024-10-21:
@PaulBouchier if you look at the latest code that I've pushed to github, you will see that I've added parameters for log_level and maxage_conn

If you want to debug try
`ros2 launch ublox_dgnss ntrip_client.launch.py use_https:=false host:=rtk2go.com port:=2101 mountpoint:=VN1 username:=paul.bouchier-at-gmail-d-com password:=Unused log_level:=DEBUG`
```
ros2 launch ublox_dgnss ntrip_client.launch.py use_https:=false host:=rtk2go.com port:=2101 mountpoint:=VN1 username:=paul.bouchier-at-gmail-d-com password:=Unused log_level:=DEBUG
[INFO] [launch]: All log files can be found below /home/nick/.ros/log/2024-10-22-07-20-46-764420-asuspn1-58272
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [58287]
[component_container_mt-1] [DEBUG] [1729545647.010712464] [rclcpp]: signal handler installed
[component_container_mt-1] [DEBUG] [1729545647.010762127] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[component_container_mt-1] [DEBUG] [1729545647.010813339] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[component_container_mt-1] [DEBUG] [1729545647.010879007] [rcl]: Initializing node 'ComponentManager' in namespace ''
[component_container_mt-1] [DEBUG] [1729545647.010894835] [rcl]: Using domain ID of '7'
[component_container_mt-1] [DEBUG] [1729545647.020384794] [rcl]: Initializing publisher for topic name '/rosout'
[component_container_mt-1] [DEBUG] [1729545647.020418778] [rcl]: Expanded and remapped topic name '/rosout'
[component_container_mt-1] [DEBUG] [1729545647.022080074] [rcl]: Publisher initialized
[component_container_mt-1] [DEBUG] [1729545647.022097613] [rcl]: Node initialized
[component_container_mt-1] [DEBUG] [1729545647.022226666] [rcl]: Initializing subscription for topic name '/parameter_events'
[component_container_mt-1] [DEBUG] [1729545647.022234936] [rcl]: Expanded and remapped topic name '/parameter_events'
[component_container_mt-1] [DEBUG] [1729545647.024026348] [rcl]: Subscription initialized
[component_container_mt-1] [DEBUG] [1729545647.024124593] [rcl]: Initializing service for service name '/ntrip_client_container/get_type_description'
[component_container_mt-1] [DEBUG] [1729545647.024136119] [rcl]: Expanded and remapped service name '/ntrip_client_container/get_type_description'
[component_container_mt-1] [DEBUG] [1729545647.024641307] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container_mt-1] [DEBUG] [1729545647.024651541] [rmw_fastrtps_cpp]: Sub Topic rq/ntrip_client_container/get_type_descriptionRequest
[component_container_mt-1] [DEBUG] [1729545647.024655255] [rmw_fastrtps_cpp]: Pub Topic rr/ntrip_client_container/get_type_descriptionReply
[component_container_mt-1] [DEBUG] [1729545647.024658396] [rmw_fastrtps_cpp]: ***********
[component_container_mt-1] [DEBUG] [1729545647.024787910] [rcl]: Service initialized
[component_container_mt-1] [DEBUG] [1729545647.024880057] [rcl]: Initializing service for service name '~/_container/load_node'
[component_container_mt-1] [DEBUG] [1729545647.024888630] [rcl]: Expanded and remapped service name '/ntrip_client_container/_container/load_node'
[component_container_mt-1] [DEBUG] [1729545647.025568801] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container_mt-1] [DEBUG] [1729545647.025584370] [rmw_fastrtps_cpp]: Sub Topic rq/ntrip_client_container/_container/load_nodeRequest
[component_container_mt-1] [DEBUG] [1729545647.025592141] [rmw_fastrtps_cpp]: Pub Topic rr/ntrip_client_container/_container/load_nodeReply
[component_container_mt-1] [DEBUG] [1729545647.025597870] [rmw_fastrtps_cpp]: ***********
[component_container_mt-1] [DEBUG] [1729545647.025747843] [rcl]: Service initialized
[component_container_mt-1] [DEBUG] [1729545647.025770644] [rcl]: Initializing service for service name '~/_container/unload_node'
[component_container_mt-1] [DEBUG] [1729545647.025787029] [rcl]: Expanded and remapped service name '/ntrip_client_container/_container/unload_node'
[component_container_mt-1] [DEBUG] [1729545647.026170692] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container_mt-1] [DEBUG] [1729545647.026180309] [rmw_fastrtps_cpp]: Sub Topic rq/ntrip_client_container/_container/unload_nodeRequest
[component_container_mt-1] [DEBUG] [1729545647.026183822] [rmw_fastrtps_cpp]: Pub Topic rr/ntrip_client_container/_container/unload_nodeReply
[component_container_mt-1] [DEBUG] [1729545647.026186877] [rmw_fastrtps_cpp]: ***********
[component_container_mt-1] [DEBUG] [1729545647.026267205] [rcl]: Service initialized
[component_container_mt-1] [DEBUG] [1729545647.026285527] [rcl]: Initializing service for service name '~/_container/list_nodes'
[component_container_mt-1] [DEBUG] [1729545647.026294078] [rcl]: Expanded and remapped service name '/ntrip_client_container/_container/list_nodes'
[component_container_mt-1] [DEBUG] [1729545647.026595766] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container_mt-1] [DEBUG] [1729545647.026603982] [rmw_fastrtps_cpp]: Sub Topic rq/ntrip_client_container/_container/list_nodesRequest
[component_container_mt-1] [DEBUG] [1729545647.026607339] [rmw_fastrtps_cpp]: Pub Topic rr/ntrip_client_container/_container/list_nodesReply
[component_container_mt-1] [DEBUG] [1729545647.026610462] [rmw_fastrtps_cpp]: ***********
[... cut ...]
Had it sitting here watching it publish the ntrip rtcm messages with `ros2 topic echo /ntrip_client/rtcm --no-arr`

There is not much more that can be done with the code at this point in time besides tweaking some of the other variables. I'll leave that to you if it adds value to your situation. Welcome to send through a PR with any other changes.

Am closing this issue as I believe now you can tweak how often stale connection with attempt to reconnect if you loose network connectivity with rtk2go.

closed 2024-10-21 by hortovanyi

hortovanyi, 2024-10-23:
@PaulBouchier added a new issue #29 for overriding QoS profiles - modified the code to add a qos override for the publisher and subscribers but haven't tested creating a launch file overriding 

Also thinking about rtk2go, have you investigated Point Perfect and SPARTN protocol?  ... initial changes have been created such that you can see if SPARTN is working (look at UBX_RXM_COR message - has both NTRIP and SPARTN)... however more work needs to be done on SPARTNKEYS as well as receiving SPARTN messages similar to the NTRIP client and pushing them to the UBLOX_DGNSS and out to the device. SPARTN messages can be received I believe either via L1 Band or via the internet.

Point Perfect with SPARTN would make using rtk2go obsolete.

PaulBouchier, 2024-10-23:
Thank you very much for your great support, and the parameter-add for
logging level. I forked your repo and am using it in my development so I
can better look into any concerns. And thanks for the info about Jazzy
supporting qos modifications - that's good to know and is something to
weigh in considering when to advance my ROS version.

I had not investigated Point Perfect - thanks for the pointer. At $42/mo
they're certainly reasonably priced for a commercial application. I'm a
hobbyist and that's more than I want to spend that way, and rtk2go works
well enough. If I find anything about why ntrip_client doesn't work with
rtk2go I'll let you know. However, one of my Dallas Personal Robotics Group
friends reported that ros-humble-ntrip-client will get you put in the
rtk2go sandbox too, if left on overnight, because it requests at 10Hz and
they only want max 1 Hz requests.

Best regards

Paul Bouchier

hortovanyi, 2024-10-23:
When using libcurl for ntrip it's a continuous stream of messages that never ends. I had to add in a way to check if it should exit the thread, so after every 10, it exits and then reconnects. That's the new connection attempts. The error just before is expected - only way to tell libcurl to return control.

So I'm not sure about the Hz as it's driven by the ntrip castor itself. Suspect it's the reconnections - you might get leverage changing it to 20. Just means if you try to terminate the node, it might need to wait longer before the thread stops.

Took a little bit of effort to get this working at first. I still think most when doing it in c/c++ use libcurl.

You could just use the command line version of libcurl and create a shell script around it, if it fails logging appropriate messages to test out different settings. 

I'm surprised the Texas government hasn't set a service up??

PaulBouchier, 2024-10-23:
Hmmm, thanks for the explanation. I don't know enough to respond, however
I'll go with what works. My DPRG friend is using
https://github.com/olvdhrm/RTK_GPS_NTRIP, which is the Kumar Robotics F9P
driver, the LORD/Microstrain ntrip_client (which is what's he reported as
getting him put in the sandbox after an overnight run, and also what I'm
using with your ublox_dgnss F9P driver) and one other package.

The Texas Department of Transportation has a corrections network that
covers the state, but they won't provide public access. 😫. Sucks for a
state with a population the same as Australia.

If you feel like a little light entertainment for a couple of minutes,
here's a summary video of my robot's 2nd place run in last year's DPRG
RoboColumbus contest. 😅
https://youtu.be/JCzpXuPcQ2M
The November 23 deadline is for this year's contest. I'm going back to New
Zealand to see my family in a few days so I am extremely time-constrained
for getting it working better for this year's contest.

Cheers

Paul

hortovanyi, 2024-10-26:
@PaulBouchier thanks for sharing. Looks interesting. Good luck this year with the contest :)