aussierobots/ublox_dgnss #35: UBX-RXM-SPARTN topic goes silent after realsense bringup

ARK3r, 2025-02-21:
Hardware:
- [Combo Breakout Board ZED-F9P + NEO-D9S](https://www.sparkfun.com/sparkfun-gnss-combo-breakout-zed-f9p-neo-d9s-qwiic.html)
- Realsense D435i
- Jetson orin nx 16GB

Using the combo board and bringing up the GPS after providing the Spartn keys to the ZED-F9P, very quickly spartn information messages start publishing on `/ubx_rxm_spartn` and `/ubx_nav_status`'s `diff_corr` becomes true after a while. But as soon as I bring up a realsense camera, even though `/gps/fix` is still publishing, the `/ubx_rxm_spartn` goes completely silent, and after a while `/ubx_nav_status`' diff_corr becomes false. Then bringing the camera down, `/ubx_nav_spartn` starts publishing again, and `diff_corr` becomes true a bit later.

Also:
- if I turn off any streams from the realsense, this issue does not happen.
- the `/dev/tty` address for the GPS module disappears after running ublox_dgnss.

I wonder if you have any pointers for how I can debug this.

hortovanyi, 2025-02-21:
I think this might be a power issue  (that is the camera is draining power and its reducing the power output on the other ports - I stopped using the Jetson equipment as I couldn't upgrade my Jetson TX2 to the latest OS. 

Suspect you might need another device, say a raspbery pi to run non-camera related tasks???

if on the same network with the same ROS_DOMAIN_ID https://docs.ros.org/en/jazzy/Tutorials/Advanced/Improved-Dynamic-Discovery.html#improveddynamicdiscovery you can see the messages on each device

ARK3r, 2025-02-21:
Thanks for the suggestion, I will try powering the realsense externally.

hortovanyi, 2025-02-22:
@ARK3r how did you go? I'm not sure if this is really an issue with the driver code itself?? .. if not is it ok to close this issue?

ARK3r, 2025-02-22:
You were right about the power, using an externally powered USB hub resolved the issue, thanks for the insight. You can close this!

closed 2025-02-22 by hortovanyi