aussierobots/ublox_dgnss #37: i2c support?

yishaiSilver, 2025-03-02:
Does this project support i2c? If I set up i2c communication between two f9ps for mb+rover operations with u-center, will this node overwrite that?

Thanks

hortovanyi, 2025-03-02:
If you've hard wired it. Modify the launch scripts to not set any associated parameters. The node itself won't overwrite existing parameters. If you want to set these parameters for i2c in a launch file they may need to be added to ubx_cfg_item.hpp otherwise modify in ucenter - if you save to ram they are lost of course if un-powered

hortovanyi, 2025-03-04:
@yishaiSilver do you want to add i2c related details to ubx_cfg_item.hpp? otherwise I will close this issue

yishaiSilver, 2025-03-04:
@hortovanyi Sorry. Turns out I had an F9R+F9P pair and the F9R wasn't able to interpret the F9P's RTCM 4072.0 messages. I just received a new F9R, so I should be able to get on that later today or tomorrow.

closed 2025-03-07 by hortovanyi