aussierobots/ublox_dgnss #38: Moving Base Not Working? Light not turning off?

yishaiSilver, 2025-03-07:
Hello again,

I've been trying to get this up and running just using UART2, which I know is much more standard.

I launch the ntrip client and I can see that my the RTCM corrections are streaming correctly.

Launching mb+r_base, my base successfully enters fixed mode. I can also see that it's publishing to /base/rtcm.

Launching mb+r_rover, nothing happens. I have successfully set the serial IDs, and I can see that the rover is publishing its position. 

/rover/ubx_nav_status shows a fix of type 3, but the light doesn't turn off. The accuracy seems reasonable (4.5 cm), but the heading isn't there?

Likewise,  ros2 topic echo /rover/ubx_rxm_rtcm shows nothing. I've added 
            {'CFG_MSGOUT_UBX_RXM_RTCM_USB': 0x1}, to the launch file. 

> header:
  stamp:
    sec: 1741310025
    nanosec: 69880883
  frame_id: rover
itow: 436443000
year: 2025
month: 3
day: 7
hour: 1
min: 13
sec: 45
valid_date: true
valid_time: true
fully_resolved: true
valid_mag: false
t_acc: 22
nano: -350026
gps_fix:
  fix_type: 3
gnss_fix_ok: true
diff_soln: true
psm:
  state: 0
head_veh_valid: false
carr_soln:
  status: 0
confirmed_avail: false
confirmed_date: true
confirmed_time: false
num_sv: 32
lon: [REDACTED]
lat:  [REDACTED]
height: 212409
hmsl: 245964
h_acc: 452
v_acc: 932
vel_n: -8
vel_e: 0
vel_d: -18
g_speed: 8
head_mot: 30326944
s_acc: 149
head_acc: 18000000
p_dop: 101
invalid_llh: false
head_veh: 0
mag_dec: 0
mag_acc: 0

It says the solution is correct?

> 
header:
  stamp:
    sec: 1741310531
    nanosec: 100882149
  frame_id: rover
itow: 436949000
gps_fix:
  fix_type: 3
gps_fix_ok: true
diff_soln: true
wkn_set: true
tow_set: true
diff_corr: false
carr_soln_valid: true
map_matching:
  status: 0
psm:
  state: 0
spoof_det:
  state: 2
carr_soln:
  status: 0
ttff: 879
msss: 1277879

I've also tried switching the roles of my system. There's only two options that come to mind: remaining thing I can think of is that there's a light storm going on. I don't think that's it since the stationary base station which is ~5km away gives my mb a fix. The other one is that I have an F9P-02B and an F9P-04B. But that shouldn't make a difference?

hortovanyi, 2025-03-07:
If they are 5 kms apart why are using 2iC? ... people normally have them on the same network with the same ROS_DOMAIN_ID ?? https://docs.ros.org/en/jazzy/Concepts/Intermediate/About-Domain-ID.html

Try looking at ubx_rxm_cor and the msg used. Think it needs to be 2

hortovanyi, 2025-03-07:
I'm not the expert on this configuration. However it sounds like you need to make sure the moving base is sending messages properly as well to the rover - it's my understanding that the launch files are setup assuming they are both on the same network. If you are hardwiring the two you need to adjust the configuration accordingly in the launch files and/or via u-Center

yishaiSilver, 2025-03-07:
No worries. I had it working with u-center, so I don't think it's too problematic. 

Here's one of my ubx_rxm_cor messages:

> header:
>   stamp:
>     sec: 1741380991
>     nanosec: 73474516
>   frame_id: rover
> version: 1
> ebno: 0
> status_info:
>   protocol: 1
>   err_status: 1
>   msg_used: 2
>   correction_id: 0
>   msg_type_valid: true
>   msg_sub_type_valid: false
>   msg_input_handle: true
>   msg_encrypted: 0
>   msg_decrypted: 0
> msg_type: 1124
> msg_sub_type: 0

I'm not sure if msg_sub_type_valid: false is the desired output.

Also, I tried doing everything over USB. I can get the rover to obtain a fix with /ntrip_client/rtcm, so I set up the base to output RTCM over USB and then remapped the rover's node from /ntrip_client/rtcm -> /base/rtcm. Once I've done that, I can see messages being published over /rover/ubx_rxm_rtcm (sometimes) and the rover enters a fixed mode. It does not, however, publish a  valid heading:

> h_valid: false
> carr_soln:
>   status: 2
> confirmed_avail: false
> confirmed_date: true
> confirmed_time: false
> num_sv: 26
> lon: 
> lat: 
> height: 197252
> hmsl: 230784
> h_acc: 26
> v_acc: 34
> vel_n: -3
> vel_e: -17
> vel_d: -100
> g_speed: 17
> head_mot: 7210109
> s_acc: 151
> head_acc: 18000000
> p_dop: 122
> invalid_llh: false
> head_veh: 0
> mag_dec: 0
> mag_acc: 0

Nothing seems to get published to relposned.

hortovanyi, 2025-03-07:
@bvsam any suggestions?

yishaiSilver, 2025-03-07:
Actually, I just got it to work by manually setting up the config with U-Center and then deleting all of the params specified by the launch files. Specifically, I don't think the CFG_MSGOUT_UBX_NAV_RELPOSNED_USB: 0x1 was working when it came to heading. Thanks for your help! It's very much appreciated.

closed 2025-03-07 by yishaiSilver

hortovanyi, 2025-03-07:
It should work. If you've set the parameters per the launch file. Which version of ros2 were you using? .... I've not recently published out a humble version. Just rolling and jazzy?

yishaiSilver, 2025-03-07:
I'm using iron due to some other reqs. Gonna move away soon. Sorry for the headache.

hortovanyi, 2025-03-07:
Iron is no longer supported