aussierobots/ublox_dgnss #41: Having error while trying ntrip

PARKasd, 2025-05-15:
[component_container_mt-1] * Failure writing output to destination
[component_container_mt-1] * Closing connection 0
[component_container_mt-1] [DEBUG 1747326480.098693031] [ntrip_client]: Processed desired count...  (DoStreaming() at /home/parkasd/Desktop/ublox_dgnss_ws/ublox_dgnss/ntrip_client_node/src/ntrip_client_node.cpp:207)

It seems like I'm having problem with writing files to destination but idk where is the targeted destination

PARKasd, 2025-05-15:
I just noticed that without Output it still sends messages

closed 2025-05-15 by PARKasd

hortovanyi, 2025-05-15:
it would of reconnected to the ntrip castor....  presently with libcurl need to trigger it to exit every n count as otherwise Control C woudnt work.

have thought about another way of doing this by setting a flag for it to exit on next message 

I might raise that as a separate issue

PARKasd, 2025-05-15:
Also, I'm bit confused when using NTRIP. When I obtain informatjon by using ros echo topic, It displays my F9P is using NTRIP. But, The LED on logicboard doesn't blink.

hortovanyi, 2025-05-15:
I rely on UBX_NAV_STATUS and  UBX_RXM_RTCM|COR messages along with the h_acc & v_acc fields ... you might have other things configured by default which its not receiving. I don't rely on the F9P device lights - plus I cant see them anyways